stm32-rs/stm32g0xx-hal #3: Next release

ijager, 2020-02-08:
Hi @dotcypress, Thanks for your work on this crate. It got me up te speed quickly with the `stm32g070`. 

I have been writing [crate examples](https://github.com/JitterCompany/adxl355-rs) based on this library. However it turns out I cannot publish that crate because it depends (`dev-dependencies`) on an unpublished version of this crate. 

So I guess my question is when you're planning to do a next release (0.7)? And whether you need any help to reach that target. 

You probably know this, but when doing a release, it would be nice to also `git tag` it so that there will be a neat release overview on github as well, [like this](https://github.com/stm32-rs/stm32f1xx-hal/releases)

dotcypress, 2020-02-08:
@ijager thank you for feedback. Updated version will be published after a new release of `stm32-rs` crate.

dotcypress, 2020-02-17:
@ijager `0.0.7` has been landed

closed 2020-02-17 by dotcypress