stm32-rs/stm32g0xx-hal #5: i2c infinite loop when building with optimizations

ijager, 2020-03-18:
When I build with optimizations  (1 or 2) enabled
```
[profile.dev]
opt-level = 1
```

Then i2c `read` operations can get into an infinite loop in the `busy_wait` macro. 
Bus speed = `=100khz`. 

```
// Wait until address was sent
busy_wait!(self.i2c, busy);
```

Here the intention is to wait until the peripheral is not busy anymore and then break out of the loop. 

```
} else if isr.$flag().bit_is_set() {
    break;
}
```

However, the busy flag is `1` if busy and will be cleared when not busy anymore. So the check should actually check for `bit_is_cleared()`. 

But I guess that's not true for all `$flags` that can be passed in.

Because optimization is on, the code is faster, the bit is already cleared so it ends up in an infinite loop.

closed 2020-04-30 by dotcypress

ijager, 2020-05-05:
Just tested 9697aef and it is not fixed yet. 

For now I need to add a 'timeout' to the `busy_wait` loops

```
 ($i2c:expr) => {
        let mut count = 0;
        loop {
            let isr = $i2c.isr.read();
            if isr.berr().bit_is_set() {
                $i2c.icr.write(|w| w.berrcf().set_bit());
                return Err(Error::BusError);
            } else if isr.arlo().bit_is_set() {
                $i2c.icr.write(|w| w.arlocf().set_bit());
                return Err(Error::ArbitrationLost);
            } else if isr.nackf().bit_is_set() {
                $i2c.icr.write(|w| w.nackcf().set_bit());
                return Err(Error::Nack);
            } else if isr.busy().bit_is_clear() {
                break;
            }
            count += 1;
            if count > 1000 {
                break;
            }
        }
    };
```

I'll try to look into it further soon

andresv, 2020-05-23:
Yes, I have the same problem.

At some point stuff from L0 https://github.com/stm32-rs/stm32l0xx-hal/blob/master/src/i2c.rs should be merged.
G0 and L0 are quite similar. However G0 uses modern DMAMUX.

ijager, 2020-05-24:
I was just getting ready to tackle this issue, so I did a git pull and turns out it is already fixed 😃
Thanks @andresv, I just tested it and it also works with optimizations on now

commit 9697aefb (2020-04-30): i2c fix #5