stm32-rs/stm32g0xx-hal #6: Build fails with --features stm32g081

andresv, 2020-04-30:
`cargo build --example blinky --features stm32g081`

```
error: symbol `DEVICE_PERIPHERALS` is already defined
    --> /Users/andres/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32g0-0.10.0/src/stm32g081/mod.rs:1160:1
     |
1160 | static mut DEVICE_PERIPHERALS: bool = false;
     | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

error: could not compile `stm32g0`.

To learn more, run the command again with --verbose.
```

andresv, 2020-04-30:
I compared `stm32g07x` and `stm32g081` dirs from here https://github.com/stm32-rs/stm32-rs-nightlies and they were exactly the same.

I also used this `stm32-rs-nightlies` to build examples, so `stm32g0xx-hal/Cargo.toml` was modified to:
```
#stm32g0 = "0.10.0"
stm32g0 = {git = "https://github.com/stm32-rs/stm32-rs-nightlies"}
```

andresv, 2020-04-30:
Ahaa, `default` is not working as expected. When I replaced `default = ["rt", "stm32g07x"]` with `default = ["rt", "stm32g081"]`. I was able to build blink with `cargo build --example blinky --features stm32g081`.

```
[features]
#default = ["rt", "stm32g07x"]
default = ["rt", "stm32g081"]
rt = ["stm32g0/rt"]
stm32g07x = ["stm32g0/stm32g07x"]
stm32g030 = ["stm32g0/stm32g030"]
stm32g031 = ["stm32g0/stm32g031"]
stm32g041 = ["stm32g0/stm32g041"]
stm32g081 = ["stm32g0/stm32g081"]
```

emilhoeiriis, 2020-05-11:
Issue persists

andresv, 2020-05-24:
This is now fixed. However now actual MCU must be specified to build examples `cargo build --example blinky --features stm32g081`.

closed 2020-05-24 by andresv