stm32-rs/stm32g0xx-hal #15: ADC VREFINT calibration values

ijager, 2020-06-03:
I needed to perform a supply-relative ADC measurement using the internal reference voltage. To do that you need to use the VREFINT calibration value, which is the adc value measured at 3V during production. 

VREFINT calibration value is a 16 bit value at `0x1FFF_75AA` (datasheet 3.13.2). I couldn't find it in the `pac`. 
For now I solved it like this in my application, but I think this should be part of the hal crate.

```rust
/// Returns a pointer to VREFINT_CAL, the VREFINT calibration value
#[inline(always)]
const fn vrefint_ptr() -> *const u16 {
    // DS12766 3.13.2
    0x1FFF_75AA as *const _
}

fn adc_abs_voltage(vref_val: u32, adc_val: u32) -> (u32,u32) {

    // safe because read only memory area
    let vref_cal: u32 = unsafe {
       ptr::read_volatile(vrefint_ptr())  as u32
    };

    // RM0454 14.9 Calculating the actual VDDA voltage using the internal reference voltage
    // V_DDA = 3 V x VREFINT_CAL / VREFINT_DATA
    let vdda_mv = 3000u32 * vref_cal / vref_val;

    // RM0454 14.9 Converting a supply-relative ADC measurement to an absolute voltage value
    //                   Vdda
    // VCHANNELx = --------------- × ADC_DATAx
    //                FULL_SCALE
    let adc_mv = (vdda_mv * adc_val / 4095u32); // TODO: full scale depends on config

    (vdda_mv, adc_mv)
}

let vref_val: u16 = adc.read(&mut vref).expect("adc read failed");
let adc_val: u16 = adc.read(&mut adc_pin).expect("adc read failed");
let (vdda_mv, adc_mv) = adc_abs_voltage(vref_val as u32, adc_val as u32);
```

Do you guys have opinions on how to implement this in de hal adc driver?

andresv, 2020-06-03:
I don't. Maybe some other stm32 hal already have this feature, we could copy design from there.

dotcypress, 2021-07-02:
@ijager could you please take a look at https://github.com/stm32-rs/stm32g0xx-hal/pull/63?

PR #63 (2021-07-06): ADC absolute voltage measurement
This PR contains experimental implementation of absolute ADC sampling.

Closes #15
closed 2021-07-06 by andresv