stm32-rs/stm32g0xx-hal #28: Blinky does not build

andresv, 2020-08-24:
`cargo build --example blinky --features stm32g031`

```
error: no rules expected the token `}`
   --> src/dmamux.rs:209:5
    |
154 | macro_rules! dma_mux {
    | -------------------- when calling this macro
...
209 |     },
    |     ^ no rules expected this token in macro call

error[E0412]: cannot find type `Channels` in this scope
   --> src/dmamux.rs:214:21
    |
214 |     type Channels = Channels;
    |                     ^^^^^^^^
    |
help: try
    |
214 |     type Channels = Self::Channels;
    |                     ^^^^^^^^^^^^^^
help: consider importing this struct
    |
1   | use crate::dma::Channels;
    |

error[E0422]: cannot find struct, variant or union type `Channels` in this scope
```

@ijager looks like recent dmamux modifications have broken it.

It works fine with `cargo build --example blinky --features stm32g071`

ijager, 2020-08-24:
I think there are two bugs:

- the HAL sometimes still creates 7 channels when there are only 5. This is easily fixed
- The DMAMUX peripheral is different in the SVD files for G030, G031 and G041. This needs to be fixed with svd patches in stm32-rs

Yatekii, 2020-08-24:
This will be fixed with #29.

closed 2020-08-25 by andresv