stm32-rs/stm32g0xx-hal #37: SO8 package - is it supported?

nebelgrau77, 2021-01-05:
It's not an issue but just a question: I'm thinking about getting the STM32G031J6 Discovery at some point, it's the one with the "8-legged" MCU. From the datasheet I understood that you can assign one of multiple GPIOs to the pins, e.g. PA8/PA11/PB1 to pin 5. Is that supported in the crate?

andresv, 2021-01-06:
I haven't tried that.

nebelgrau77, 2021-01-06:
OK, so I tried the following: created a project in CubeMX, assigning the GPIOs to actual chip pins, e.g. PA11 on pin 5, PA1 on pin 4 and so on. The generated project looks just like you'd have it on any other MCU, with the selected pins defined as inputs or outputs. I couldn't find any special declaration on what's used as what, so I assume (but may be totally wrong about it) that CubeMX simply takes care of it by letting you choose only one of the possible GPIOs for each physical pin. As I don't have the board, can't tell what happens if you declare two GPIOs that correspond to one physical pin.

almusil, 2021-01-19:
Those pins are bonded physically together. As stated in `Multi bonding on STM32G03xxx and STM32G04xxx small packages` [AN5096](https://www.st.com/resource/en/application_note/dm00443870.pdf):
> The configuration state of each device pad must be done carefully as there are not any design protections presentto avoid interferences from one pad to the other.

So there is nothing special needed just being cautious which pins are you using.

closed 2021-01-19 by andresv

nebelgrau77, 2021-01-19:
Great, thank you!

nebelgrau77, 2021-03-22:
I can just confirm that it works with the small package, but if you're using the Discovery kit and get it updated to the latest firmware version, st-flash will throw a chip ID error. Flashing with Black Magic Probe instead solves the problem :)