stm32-rs/stm32g0xx-hal #87: Configuring I2C pins causes them to be temporarily pulled low

davidlattimore, 2021-11-09:
Currently SDAPin and SCLPin are only implemented for pins in mode Output<OpenDrain>. Configuring the pins as open-drain prior to them being reconfigured for their alternate function results in the pins momentarily being pulled low (see scope trace below). I've found that this can confuse other devices on the bus. In my application, I'm currently working around this by using core::mem::transmute to construct a pin that thinks it's configured as open-drain, when it is in fact still in its default state. This seems to work well, but is pretty ugly.

Would you be open to implementing SDAPin and SCLPin for the pins in their default state? If so, I'm happy to send a pull request. Would you prefer that I implement for both DefaultMode and Output<OpenDrain> or just DefaultMode? Removing the implementations for Output<OpenDrain> would be a breaking change.

[image]

andresv, 2021-11-11:
@dotcypress do you have any thoughts about this?

dotcypress, 2021-11-11:
@davidlattimore Good catch!
I think we can implement SDAPin/SCLPin for pins in default mode.

I can try to do that on a weekend, but if you have something already - I will be happy to review.

PR #88 (2021-11-11): i2c: Take pins in DefaultMode rather than Output<OpenDrain>
Putting pins into open-drain mode causes them to be pulled low, which
can confuse other devices on the bus.

This is a breaking change. Calling code can be fixed by not calling
`into_open_drain_output()` on pins when setting up I2C.

Fixes #87
commit 870eaef0 (2021-11-11): i2c: Take pins in DefaultMode rather than Output<OpenDrain>

Putting pins into open-drain mode causes them to be pulled low, which
can confuse other devices on the bus.

This is a breaking change. Calling code can be fixed by not calling
`into_open_drain_output()` on pins when setting up I2C.

Fixes #87
closed 2021-11-11 by andresv