stm32-rs/stm32g0xx-hal #93: Allow output value to be set before configuring pin as an output

davidlattimore, 2021-12-21:
Somewhat related to #87 (but more general)....

Sometimes you have a pin that has an external pull-up and you'd like to configure the pin as an output (generally an open-drain output) without momentarily pulling the pin low. I'd like it if we could have an API that allowed this. Possibilities that come to mind:

1. Implement OutputPin for Analog. Downside would be that users might call set_low/high and expect an immediate effect.
1. Implement some other method on pins that does the same thing as set_low/high, but makes the intent clearer. e.g. `preset_low` / `preset_high`.
1. Add an argument to into_open_drain to specify the initial state. But this would be a breaking change.
1. Add a new API for creating an OpenDrain output that allowed specifying the initial value and make into_open_drain just be a shortcut for this.

Thoughts?

andresv, 2021-12-22:
Do you happen to know if something like this is available in some other HAL?

andresv, 2021-12-22:
Then it makes sense to do it the same way. Can you make a PR.

@dotcypress would you be happy with such change?

dotcypress, 2021-12-22:
Yes, we definitely need to fix that. Let me think, probably provided solution is fine.

PR #94 (2021-12-25): Allow setting initial state when configuring output pins
Fixes #93
commit 10aee2fb (2021-12-25): Allow setting initial state when configuring output pins

Fixes #93
closed 2021-12-25 by andresv