stm32-rs/stm32g0xx-hal #100: [dependencies.stm32g0xx-hal] produces error during cargo build "uplicate key: `stm32g0xx-hal`"

jemo07, 2022-04-03:
Hello, 

I get the following error while trying to build after just adding entries to Cargo.toml. Here is the excerpt from that file:

```
[dependencies.stm32g0xx-hal]
version = "0.1.3"
features = ["rt", "stm32g030"]

[dependencies]
embedded-hal = "0.2.7"
bit_field = "~0.10"
bme280-multibus = "0.2.1"
stm32g0xx-hal = "0.1.3"
```

Here is the outcome changing the order by placing the `[dependencies.stm32g0xx-hal]` before/after `[dependencies]` 

```
Caused by:
  duplicate key: `stm32g0xx-hal` for key `dependencies` at line 11 column 1
jemo@jemoLINUX:~/development/git/nuecleo-bme280$ cargo build
error: failed to parse manifest at `/home/jemo/development/git/nuecleo-bme280/Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  duplicate key: `stm32g0xx-hal` for key `dependencies` at line 14 column 1
jemo@jemoLINUX:~/development/git/nuecleo-bme280$ 
```

andresv, 2022-04-03:
Try like this:

```toml
[dependencies]
embedded-hal = "0.2.7"
bit_field = "~0.10"
bme280-multibus = "0.2.1"
stm32g0xx-hal = { version = "0.1.3", features = ["rt", "stm32g030"] }
```

andresv, 2022-04-03:
If you use it like this:

```toml
[dependencies.stm32g0xx-hal]
version = "0.1.3"
features = ["rt", "stm32g030"]
```

Then this must be removed, otherwise it is duplicated:
```diff
[dependencies]
embedded-hal = "0.2.7"
bit_field = "~0.10"
bme280-multibus = "0.2.1"
- stm32g0xx-hal = "0.1.3"
```

closed 2022-04-03 by andresv