stm32-rs/stm32g0xx-hal #104: Feature: fully erasable pins

jacobbarssbailey, 2022-06-14:
GPIO in this project supports partially erased pin assignments, where the pin number is erased, allowing a group of related input or output pins to be collected in an array under a single type. However it doesn't seem possible to remove the port number as well.

```
let generic_pa_pin_1 = gpioa.pa1.into_push_pull().downgrade(); // PAx<Output>
```

…but for the general use case of wanting to create an array of pins, using only pins from a single port seems rather confining.

I've noticed that [stm32f1xx-hal](https://github.com/stm32-rs/stm32f1xx-hal) supports both partially and fully erased pin (erasing the pin number and the port respectively), by adding [both Partially Erased and Fully Erased Pin types](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/gpio.rs#L83), with Partially Erased pins downgradable to Fully Erased pins:

```
let generic_pa_pin = gpioa.pa0.into_push_pull().downgrade(); // PAx<Output>
let generic_pin = generic_pa_pin.downgrade(); // PXx<Output>
```

To my (admittedly naive) eye it seems like this approach could be ported to this project without introducing any breaking changes, allowing for significantly more flexibility in managing IO pins.

If this sounds like a reasonable approach, I'd be happy to put together a pull request, but I wanted to make sure it would be a welcome and reasonable change before putting in the work.

andresv, 2022-06-14:
Yes I think it would be useful addition.

jacobbarssbailey, 2022-06-14:
Wonderful! I'll see what I can do to get a PR together

closed 2022-07-07 by dotcypress