stm32-rs/stm32g0xx-hal #110: Invalid clock configuration for apb_clk, apb_tim_clk and core_clk

keadwen, 2022-07-14:
G'day

I've identified an inconsistency in the RCC configuration.
According to [page 166/1390 RM0444 Rev 5](https://www.st.com/resource/en/reference_manual/rm0444-stm32g0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), the apb_clk is not driven by sys_freq, but ahb_freq.

In the further review of the RCC configuration freeze, I've noticed 3 separate inconsistencies with the RM0444 and STM32CubeMX Clock Configuration section:

1. apb_clk input is sys_freq ([code reference](https://github.com/stm32-rs/stm32g0xx-hal/blob/main/src/rcc/mod.rs#L142-L148)), when it should be ahb_clk.
2. apb_tim_clk is always set as /2 ([code reference](https://github.com/stm32-rs/stm32g0xx-hal/blob/main/src/rcc/mod.rs#L142-L148)), when it should be:
    - x1 when apb_psc is x1
    - x2 when apb_psc is x2 or greater
3. core_clk is ahb_freq / 8 ([code reference](https://github.com/stm32-rs/stm32g0xx-hal/blob/main/src/rcc/mod.rs#L180)), when it should be either /1 or /8 (while /8 is default).

Image reference from STM32CubeMX for STM32G071GBU6
[image]

Thanks,
Jakub

keadwen, 2022-07-19:
Tiny correction to point 2.

The logic in the code was correct. Only the reference clock was invalid.
If a prescaler is set NotDivided then we have /1 (captured in _ match).
Otherwise, we do have /2 (captured in other matches).

andresv, 2022-07-22:
@keadwen thanks for finding this out. I'll double check it later.

PR #111 (2022-08-01): Correct the APB frequency calculation
For more details, refer to https://github.com/stm32-rs/stm32g0xx-hal/issues/110

Resolves #110
closed 2022-08-01 by andresv