stm32-rs/stm32g0xx-hal #127: I think FDCAN is missing?

brandonros, 2022-11-07:
Sorry, not trying to be annoying.

https://github.com/stm32-rs/stm32g4xx-hal/blob/16d6ae2a4b7d38805fbe614d6494ddd2a83429c8/src/fdcan.rs

just not sure if that is supposed to be ported over to this repo?

mattico, 2022-11-08:
Yep! There are many things missing. The device family HALs are mostly separate projects run by mostly separate people. If you're looking to add FDCAN support, copying the most complete version from the most similar device can be a good place to start.

The [stm32-rs/stm32-rs ](https://github.com/stm32-rs/stm32-rs) data files can help to find which devices have similar peripherals, or [the peripheral map from embassy-rs](https://github.com/embassy-rs/stm32-data/blob/3bd8cf74e69eb7c6d905a5ba28bcaa3fffc12ac1/stm32data/__main__.py#L84) though it looks like they haven't implemented FDCAN yet. And in this case the STM32G4's FDCAN registers are probably very similar.

Even if the G4's FDCAN peripheral is identical to the G0 peripheral there are usually different bugs in ST's SVD files, which are used to generate the register access HAL, which will need to be fixed.

brandonros, 2022-11-08:
Appreciate the insight.

Not to add more fire to the flame but it seems that… the fdcan driver itself is in the process of being copied/moved out of the stm32gfxxx repos and moved into its own at https://github.com/stm32-rs/fdcan… maybe? I’m not sure

Then that was was supposed to implement embedded-can traits for embedded-hal, but that’s commented out (just putting that here for anybody else that tries to go down this rabbit hole)

I’m sure the community will catch up/clean up support for these chips soon enough, thank you for everything this team does for the community!

mattico, 2022-11-10:
Didn't intend to sound testy, just wanted to add some info for if you needed to get FDCAN working.