stm32-rs/stm32g0xx-hal #143: Spelling error

dauntown, 2024-01-15:
```lang=rust
pub trait DmaMode<ADC> {
    /// Error type returned by ADC methods
    type Error;
    fn dma_enable(&mut self, enable: bool);
    fn dma_circualr_mode(&mut self, enable: bool);   // <=======
}
```
Here DmaMode contains the method `dma_circualr_mode()`. Please correct it to `dma_circular_mode()`. The name error doesn't inspire confidence it will in fact work. 

As seen on https://github.com/stm32-rs/stm32g0xx-hal/blob/e6145ef83c639e736556fb4775f349aef7100ad6/src/analog/adc.rs#L374C16-L374C16

Disasm, 2024-01-15:
Thanks! Would you like to create a PR to fix the typo?

dauntown, 2024-01-15:
No, the honor, and the breaking change, is all yours.

commit 6d4488f2 (2024-12-16): [Breaking change] Fix spelling error. Closes #143
closed 2024-12-16 by dotcypress